espterm/espterm-firmware #65: double size lines

MightyPork, 2017-08-19:
http://vt100.net/docs/vt102-ug/chapter5.html#S5.5.2.19

cpsdqs, 2017-11-09:
The backend code shouldn't become that complicated—xterm just stops moving the cursor when it reaches the end of the screen, it doesn't even wrap or delete the now invisible screen data. Could probably have an array of 25 uint8's containing flags for each line and just send that as another topic to the client. And rendering it would then be pretty simple

MightyPork, 2017-11-09:
mm I think it won't be so bad indeed. If you add it to the front-end, I'll add the backend support

The topic could be more compact, e.g. (only non-normal lines):
(count)(line)(line)(...

line = attrs|(line<<3)

with attrs being 3 bits: double width, upper of dual height, lower of dual height

cpsdqs, 2017-11-09:
https://github.com/espterm/espterm-front-end/commit/f61d861883c59a6a79efd740c0a3ed0784d51ebc
[image]

closed 2017-11-12 by MightyPork