espterm/espterm-firmware #74: Allow default background color being off-scale

MightyPork, 2017-08-20:
however since colors are 4-bit, we have to add this as an attribute to the attrs bitfield, and take care of it when serializing screen to JS string.

one extra bit will fit into Encode2B safely, or we can finally move to binary messages in socket (but there may be a problem with the init XHR?)

MightyPork, 2017-09-12:
alternatively we could give up full rgb hex and just pick from the 256 colors, that makes it just one byte that's a bit easier to handle

cpsdqs, 2017-09-14:
the attributes byte could also be extended like UTF-8 to have a bit that, when set, extends it to two bytes

MightyPork, 2017-09-14:
the issue is with back-end storage, not transport encoding.

With 80x25 that is 2000 bytes for each byte added to the cell struct

Currently, that is:

- 1 byte - character or utf-8 look-up table reference
- 1 byte - fg 0-255
- 1 byte - bg 0-255
- 1 byte - attribute bits

I'd love to cut it further, but have not devised a way to do that yet. your idea with "attribute ranges" would blow up with something like the rainbow demo...

cpsdqs, 2017-09-14:
Yeah that'd be like trying to compress RGB noise losslessly

MightyPork, 2017-09-24:
This is now implemented, details are in #191.

closed 2017-09-24 by MightyPork