espterm/espterm-firmware #121: [canvas] Overflowing characters are clipped by redrawn backgrounds

cpsdqs, 2017-09-09:
- when redrawing, check for large characters and set needsUpdate for them too

cpsdqs, 2017-09-10:
Might've been fixed with https://github.com/espterm/espterm-front-end/commit/4731a259dd68cc20d486e8e1a751a7d9d02fa3c8 (please check)

MightyPork, 2017-09-10:
the overlaying glitch is fixed, but there's at least one situation where the erase glitch remains:

I tested it with `🚉` 

[image] 
[image] 
[image] 
[image]

it happens both on the same line and line above and below.

does not happen when it's neighbored by a non-blank character from the side closer to cursor

cpsdqs, 2017-09-10:
Try now; you can see which characters are being updated by doing `Screen.window.debug = true` (one of the few benefits of having everything in global scope)

MightyPork, 2017-09-10:
it's better! 

there is this small bug left, but i don't think it's worth wasting time on

(moved from the smiley leftwards)
[image]

btw is it possible to enable sourcemaps for the babel minify plugin?

cpsdqs, 2017-09-10:
Well, usually you'd be using webpack (which adds a few dozen bytes (obviously)) or something like gulp, which would do it for you. Though the `--source-maps` option does seem to work with the babel compiler, but not with the standalone minifier.

cpsdqs, 2017-09-10:
so _is_ this fixed now? Because basically what I've done is just increase the radius of redrawn cells each time

MightyPork, 2017-09-10:
not really, but it's better than it was before.

eg this bug is still there - 😛 with 1 space in between
[image]

MightyPork, 2017-09-10:
is clipping expensive? because you could just set clip region and redraw all surrounding wide cells after redrawing the insides

MightyPork, 2017-09-10:
oh, and that character disappearing bug from before with the train station is still there, but it's now y+2 and y-1 lines lol

but it's ok, printing emojis is not really a v important feature imo

cpsdqs, 2017-09-10:
I don't get how your image with 😛 came to be (unless you disabled debug mode halfway through) because as far as I can see the entire region is being redrawn correctly. And no, clipping shouldn't be expensive; I guess I could rewrite that part some time soon

MightyPork, 2017-09-10:
Yeah I had debugging enabled before and switched it off cause that magenta overlay was obscuring my experiments

cpsdqs, 2017-09-10:
...I would rewrite the debug mode to use a second canvas that shows the updates, but wasn't this whole thing under some sort of size limit or something?

MightyPork, 2017-09-10:
I have about 200 more kilobytes of ROM to spare, but if it could be removed
from the production build somehow that would be better ofc.

The bigger issue is RAM, which is elegantly worked around by throwing the
hard work at the browser

cpsdqs, 2017-09-10:
oh wow you could at least fit half a percent of the node_modules folder in there
and also use webpack, or babel in case you do want to support the Android 4.4.4 or less browser

cpsdqs, 2017-09-11:
So as it turns out, clipping _is_ rather expensive, at least with the current algorithm (which sweeps each line and then adds horizontal regions to the mask; the fewer regions, the better) with 156ms for a full redraw vs 19ms without. Thus I disabled it by default, but it can be enabled by setting `Screen.window.graphics` to `1` (or higher)

https://github.com/espterm/espterm-front-end/commit/de897f013c9259a59e0e357a8c9ea3203c384763

MightyPork, 2017-09-11:
Works great with graphics=2, a full redraw time here is max 46 ms in my Chrome.

worse than 19, but pretty good.
maybe we can enable/disable it dynamically based on the presence of wide characters on the screen (most of the time that won't be the case, and if they are there, it's usually some static UI like a meteo-station panel with weather symbols)

closed 2017-09-11 by MightyPork