espterm/espterm-firmware #125: Debug toolbar

MightyPork, 2017-09-09:
Inspired by the Status Line feature of some VT models, would show current screen and cursor state, selected modes and allow their manipulation.

cpsdqs, 2017-09-09:
When using the canvas renderer, maybe also include a visual debug mode showing screen updates or overlaying each cell's attributes.

cpsdqs, 2017-10-08:
As the last remaining item; what does “make it pretty” entail?

MightyPork, 2017-10-08:
wellll that's a little abstract xD
originally i thought it'd be more visual, it's mainly for the user to debug their application code, not us.

i should add some more stuff to the Debug topic tbh, like active cursor colors

cpsdqs, 2017-10-08:
ooooh yeah I kind of used it for debugging myself, so that's what I thought it was for xD
Yeah then there should probably be a button for _enabling_ draw debugging and it should probably be off by default

cpsdqs, 2017-10-08:
so basically kind of like this?
[image]

MightyPork, 2017-10-08:
oo if you could do that, this is kinda what i imagined yeah

there should also be a toggle for Input Stream Debug¹ - sent to server (need to add this to the update word), and a button to reset the screen/parser - likewise sent to server. And a display of selected colors and other stuff that are atm not in the update word but could be...
¹
[image]

cpsdqs, 2017-10-08:
uhh I think it would be nicer if you could see both the input stream and the output, i.e. send that in a separate update topic

MightyPork, 2017-10-08:
hnn well this is already implemented (it uses the screen for display). can't really store everything and send it to the front-end, maybe "last 10 bytes" or something

MightyPork, 2017-10-08:
but like, if you add that, nothing against it, i might add back-end support later..

cpsdqs, 2017-10-08:
yeah I mean you could just send the raw data through the websocket and let the javascripts handle it

MightyPork, 2017-10-08:
the socket is quite slow, idk how well that'd work.. plus would need to be binary socket, bc there can be invalid utf8 pieces in it

MightyPork, 2017-10-08:
hmm also you still don't have display and data at the same time, unless you'd open two sockets in parallel

MightyPork, 2017-10-08:
but sending last n bytes but display in the debugbar is a neat idea, i kinda want that

there's already such a buffer in the back-end for showing context when a parse fails

cpsdqs, 2017-10-08:
okay then I shall do a debugger makeover after I finish the more hierarchical and less interwoven ScreenLayout thing

cpsdqs, 2017-10-27:
Merged `fancy-debug` into `work`: https://github.com/espterm/espterm-front-end/commit/78d32f7fdcf40430287cfed0bdaeb032560de2a9

closed 2017-10-27 by MightyPork