espterm/espterm-firmware #129: Line and Bar cursors don't clear cell when inactive, leave traces

MightyPork, 2017-09-09:
[image]

(Cursor style is now configurable in terminal settings and via command, sent as bits 11..9 of attribs word)

cpsdqs, 2017-09-09:
uhh I just pushed a commit that also added the new cursor styles and now it's got code from both. I did fix that issue in mine though, so I guess I'll use that?

MightyPork, 2017-09-09:
uh. I guess?

but I think this is easier to read

```js
    let cursorStyle = (attributes >> 9) & 0x07;
    // 0 - Block blink, 2 - Block steady
    // 3 - Under blink, 4 - Under steady
    // 5 - I-bar blink, 6 - I-bar steady
    this.cursor.style = cursorStyle<3?'block':cursorStyle<5?'line':'bar';
    this.cursor.blinkEnable = cursorStyle === 0 || cursorStyle === 3 || cursorStyle === 5;
```

it's never 1, that's taken care of server-side.

MightyPork, 2017-09-09:
the redraw fix works great btw!

cpsdqs, 2017-09-09:
oh, then this is what'd be left of my decoder
```js
    let cursorShape = (attributes >> 9 & 0x07);
    let cursorStyle = cursorShape >> 1;        // I think this part makes more sense than < 3 and < 5
    let cursorBlinking = !(cursorShape & 1);

    if (cursorStyle === 0) this.cursor.style = 'block';
    else if (cursorStyle === 1) this.cursor.style = 'line';
    else if (cursorStyle === 2) this.cursor.style = 'bar';

    if (this.cursor.blinking !== cursorBlinking) {
      this.cursor.blinking = cursorBlinking;
      this.resetCursorBlink();
    }
```

MightyPork, 2017-09-09:
yeah... with some mental gymnastics I can see how that works. can you keep the legend comment tho?

cpsdqs, 2017-09-09:
yeah, but I see that 1 is missing and you're using < 3; are there 6 (as you said above) or 7 (as in the code) cursor styles?

MightyPork, 2017-09-09:
numbers 1 and 7 are unused, so there are 6, blinky and steady variant of the three

MightyPork, 2017-09-09:
however this gives me an idea to save one bit in the word - if I use number 1 or 7 to mean "cursor is hidden", then the bit number 0 of the attribs word will be free for some future use

MightyPork, 2017-09-09:
dw about that, I'm not doing this now xD
but bits may get scarce later

closed 2017-09-09 by MightyPork