espterm/espterm-firmware #130: [meta] Add code style spec

cpsdqs, 2017-09-09:
Either a very exhaustive text document or with e.g. an `eslintrc.json` in the project root

MightyPork, 2017-09-09:
hmm if phpstorm / webstorm can read eslintrc.json, then why not

cpsdqs, 2017-09-09:
sure can https://www.jetbrains.com/help/phpstorm/eslint.html
and while I adhere to https://standardjs.com, I feel like this does not; so you should probably do it

MightyPork, 2017-09-09:
I'd rather have semicolons than mysterious syntax errors after minifying it.. 

also eg. `456 & 1 << 2` - without parens like you put it there, I don't think that's very clear and I wouldn't dare guess what precedence will be used (eg it's probably swapped in php or c etc)

cpsdqs, 2017-09-10:
- Any minifier that doesn't add invisible semicolons is broken
- I don't think php should be considered a valid frame of reference for operators xD
- I guess that is rather unclear; I'll keep that in mind

MightyPork, 2017-09-10:
mk but also I hope your best™ syntax style allows for else to go on a new line, because that makes code so much easier to read

cpsdqs, 2017-09-10:
else on a _new line_? Scandalous. And no it's the regular old `} else {` style -- I mean, for the record, you could just copy their eslintrc and modify that
...but for all intents and purposes I won't touch the code style

MightyPork, 2017-09-10:
yeah eg

```js
     if (fitIntoWidth && fitIntoHeight) {
        if (realWidth > fitIntoWidth || realHeight > fitIntoHeight) {
          let terminalAspect = realWidth / realHeight
          let fitAspect = fitIntoWidth / fitIntoHeight

          if (terminalAspect < fitAspect) {
            // align heights
            realHeight = fitIntoHeight
            realWidth = realHeight * terminalAspect
          } else {
            // align widths
            realWidth = fitIntoWidth
            realHeight = realWidth / terminalAspect
          }
        }
      } else if (fitIntoWidth && realWidth > fitIntoWidth) {
        realHeight = fitIntoWidth / (realWidth / realHeight)
        realWidth = fitIntoWidth
      } else if (fitIntoHeight && realHeight > fitIntoHeight) {
        realWidth = fitIntoHeight * (realWidth / realHeight)
        realHeight = fitIntoHeight
      }
```

vs

```js
     if (fitIntoWidth && fitIntoHeight) {
        if (realWidth > fitIntoWidth || realHeight > fitIntoHeight) {
          let terminalAspect = realWidth / realHeight
          let fitAspect = fitIntoWidth / fitIntoHeight

          if (terminalAspect < fitAspect) {
            // align heights
            realHeight = fitIntoHeight
            realWidth = realHeight * terminalAspect
          } 
          else {
            // align widths
            realWidth = fitIntoWidth
            realHeight = realWidth / terminalAspect
          }
        }
      } 
      else if (fitIntoWidth && realWidth > fitIntoWidth) {
        realHeight = fitIntoWidth / (realWidth / realHeight)
        realWidth = fitIntoWidth
      } 
      else if (fitIntoHeight && realHeight > fitIntoHeight) {
        realWidth = fitIntoHeight * (realWidth / realHeight)
        realHeight = fitIntoHeight
      }
```

cpsdqs, 2017-09-10:
I guess you can see the `if`s more easily
...just don't make operators have zero spacing like `1+1` xD

MightyPork, 2017-09-10:
yea that happens when im too focused on the algo to remember formatting exists xD

MightyPork, 2017-09-10:
okay there's now `.eslintrc` from standard style with a few exceptions that were giving me false positives (like it thinking a global is undefined), allowed snake_case and ==

also [this delicious hack](https://github.com/espterm/espterm-front-end/blob/canvas/build.sh#L5) was needed to make it work without semicolons

closed 2017-09-10 by MightyPork