espterm/espterm-firmware #131: Clean up the front-end directory structure

MightyPork, 2017-09-10:
After canvas merge, to avoid conflicts.

- move js/ and css/ into the out/ folder
- rename jssrc to js/ or javascript/
- have the build script also copy images and other files
- modify the firmware build script to retrieve the static files from here

cpsdqs, 2017-09-10:
also, the build script doesn't work if the enclosing directory doesn't have an html folder.

MightyPork, 2017-09-10:
oh it should, I changed that (maybe it didn't get into the canvas branch ...)

cpsdqs, 2017-09-11:
Can this be done now?

Also, copy img/ to out/ and set relative paths in _head.php instead of `/`css/app.css or something

MightyPork, 2017-09-11:
Yeah if you want to try it, go ahead. I was planning to do that tonight (was meant to work, but ehhh ... not happening )

relative paths aren't possible because on the esp they are "fancy urls" and it wouldn't work.

with the php server running, those work fine for testing...

cpsdqs, 2017-09-11:
espterm-front-end done

MightyPork, 2017-09-11:
oh i should update the backend, huuuh

closed 2017-09-11 by MightyPork