espterm/espterm-firmware #132: Text editing shortcuts

cpsdqs, 2017-09-10:
Text editing shortcuts such as

- ⌥← to go back a word (`^[b`)
- ⌥→ to go forward one word (`^[f`)
- ⌘← to go to the beginning of a line (`^A`)
- ⌘→ to go to the end of a line (`^E`)
- ⌥⌫ to delete a word (`^W`, I think)
- ⌘⌫ to delete to the beginning of a line (possibly `^U`)

are missing. I also don't know what their non-macOS counterparts are, though I suspect this'll require knowing whether or not the user agent is on macOS/iOS-with-a-hardware-keyboard.

MightyPork, 2017-09-10:
can you try to grab what sequences those are meant to send?

I'd suggest you run hexcat (https://github.com/MightyPork/smolcat/blob/master/hexcat.c), but that won't compile on mac, so try tome node hack instead i guess

MightyPork, 2017-09-10:
also i don't think they have any non-mac counterparts.

^A and ^E seem to work with nano, that's it

cpsdqs, 2017-09-10:
Not even CSI b and CSI f? Also, delete word sends \x17 and delete line sends \x15

MightyPork, 2017-09-10:
CSI b should mean Repeat Last Char, CSI f is an obsolete form of CSI H and means Absolute Position. Neither does anything in editors (but maybe ncurses doesnt like me typing it manually)

cpsdqs, 2017-09-10:
yeah sorry I meant ESC instead of CSI

MightyPork, 2017-09-10:
nah doesnt seem to do anything

MightyPork, 2017-09-10:
ctrl+left, right works to move one word though

cpsdqs, 2017-09-10:
what's ctrl+left/right sending? For me, ESC b toggles backup files and ESC f toggles multiple file buffers in nano. Vim supports them though (naturally)

MightyPork, 2017-09-10:
```
ESC [ 1 ; 5 D ctrl left
ESC [ 1 ; 5 C ctrl right
ESC [ 1 ; 5 A ctrl up
ESC [ 1 ; 5 B ctrl down
ESC O H home
ESC O F end
ESC [ 5 ~ pg up
ESC [ 6 ~ pg dn
```

can't seem to get vim to do anything because it grabs the esc and thinks it's normal esc to change mode

cpsdqs, 2017-09-10:
Huh. I guess you can add those, if they work? Do note though that you can't use ctrl+left/up/right/down on macOS because those are reserved for Mission Control by default

closed 2017-09-10 by MightyPork