espterm/espterm-firmware #136: WIFI issues

scargill, 2017-09-10:
Not working for me - I've told it to scan for my WIFI network (running on it's own internal access point on my Android phone Chrome browser) and it can't see ANY of my access points.

Also I can't see how to turn the terminal into full screen mode without all the controls.

MightyPork, 2017-09-10:
I'm sorry, but it works fine here, I'm scanning WiFi just now and see a bunch of access points from neighboring homes. So not sure what to tell you to try... 

Fullscreen mode is not a feature we currently have, but it might be added later (#139). We're in the process of re-writing the terminal front-end to work better on mobile, in Internet Explorer and Safari, and this should hopefully be possible then as a side effect.

The current latest release lets you hide the buttons and menu links under the screen, at least.

MightyPork, 2017-09-11:
I just tried it again with a completely clean ESP flash (testing ESP-12 and ESP-01), and both worked fine. Maybe something improved since the latest release?

I wanted to wait for all the new features to be complete before making a release, but here it is - a beta pre-release of 1.1. Please let me know if the problem is fixed:

https://github.com/espterm/espterm-firmware/releases/tag/1.1.0-beta

MightyPork, 2017-09-11:
okay so, there was a JavaScript bug. I need to test more thoroughly next time...

[image]

closed 2017-09-11 by MightyPork

MightyPork, 2017-09-11:
in case you still want 1.0, I just made a patch 1.0.2 that should address this:
https://github.com/espterm/espterm-firmware/releases/tag/1.0.2

MightyPork, 2017-09-11:
> Yes, that works a treat – but now…
>
>
>
> I’m all set up to go – got it on my browser in my PC – got the little
> terminal window..
>
>
>
> Config - set to 80 by 40 – apply validation errors
>
>
>
> I manage to set the width to 80 – and that was fine – but height would not
> increase..
>
>
>
> But then when I got to the terminal I could see the improved width – but
> then going back to config – nothing – the unit had failed…
>
>
>
> I hit reset and all was well, but any attempt to set the height over 15 –
> failed.

The screen size can be at most 2000 characters large - that is 80x25. 80x25
works, I use that for development.
I guess this should be indicated on the config page.

there is very little RAM in the ESP (now about 26kB, after a user connects).

This was much worse in v. 1.0, it had about 15 kB. I managed to compress
things a little,
so I may raise the limit yet. If you build from source, there's a constant
for this in `screen.h`.

What happened with the messed up settings:
When you submit, it goes through the parameters sent and applies them to
the Live Config struct.
If all is well, this Live Config is then copied to Flash. However, if you
put in invalid values, it fails and
nothing is saved. I suspect it changed some parameters, but ultimately
failed because of the screen size limit.
A restart cleared the Live Config and all reverted to the previous state.

(This may need some work, I know what values to put there so this case
wasn't tested very much)

scargill, 2017-09-11:
Something else-  might be me mis-understanding things..

 

If I turn local echo on – and send “hello” followed by a carriage return – that’s what I get – a carriage return – no line feed.  If this is normal behaviour – could you possibly add  an option to do the line feed automatically – that’s what I’d expect from the likes of Putty….

 

 

Pete.

 

https://github.com/espterm/espterm-firmware/releases/tag/1.0.2

MightyPork, 2017-09-11:
This option is there :D

Yes, this is the normal and expected behavior - enter sends code 13.
Ctrl+Enter sends 13+10.

If you go to the terminal config, it's about 3 options above Local Echo in
the expert config section.

scargill, 2017-09-11:
Ah of course- RAM…. Hmm, maybe ESP32 would have been a better option given that they are no coming down in price rapidly – Banggood had them for under £4 over the weekend – you could then do stuff like 132x50  - nice.

 

Still – great work!!!

scargill, 2017-09-11:
Magic.

MightyPork, 2017-09-11:
> hSIpgaJpZM4PSeO6> .

You're right with the ESP32, I already have one in the drawer for testing!
Just haven't gotten around to try anything with it yet.

This was developed originally for use in a uni course, so the requirement
was to be super cheap.. students tend to abuse things like this :/

There needs to be some headroom for the websocket and HTTP server
 e.t.c., when someone like Guenter opens 3 browser windows at once to
"stress test", it gets a bit tight.

I think we could go up to your 80x40 with the recent RAM improvements
though... I'll do some testing before a full v. 1.1 release

scargill, 2017-09-11:
Erm…

 

I was trying to send escape characters to the terminal and somehow it has stopped accepting CRLF – it just stays on the same line….  I’ve even rebooted the board and it is staying the same despite talking to it via 2 different serial terminals.   

 

Something seems to be “stuck”.

 

Pete.

MightyPork, 2017-09-11:
You didn't give me much to guess from for the cause. I've never had this
happen...

If it's persistent, that can hardly by something you sent to it that caused
this error.
No options set by commands are persisted to Flash

If you connect a usb-serial Rx to GPIO2, you can read debug messages.
Most debugging in the beta build is disabled, but errors should appear, if
there are any. Maybe it helps ...