espterm/espterm-firmware #139: Fullscreen mode

MightyPork, 2017-09-10:
Maybe we can abuse the zoom feature to make it fill a larger window too?

cpsdqs, 2017-09-11:
We can, but where would you put the button?

MightyPork, 2017-09-11:
I would hide everything but the actual screen I guess? Also the blue outline possibly

(btw did you notice there's a strange transparent padding at the bottom of the canvas?)

I think there would have to be a special shortcut for this, a config option, and a DECSET command (`\e[?802h` and `\e[?802l`, perhaps).

Button would be nice, but it really gets cluttered in the bottom menu. Maybe we can hide Help and About on mobile, the view kinda sucks anyway without a desktop browser and nobody will use phone for development.

cpsdqs, 2017-09-11:
Added in https://github.com/espterm/espterm-front-end/commit/2c3f081d994b6658f1f24bcc03b2e2ddd181533d, might be really buggy in some browsers, nothing I can really do about it

MightyPork, 2017-09-11:
was editing the same file :C
check if I didn't mess something up with the merge

MightyPork, 2017-09-11:
okay i'm gonna close this issue for now, seems adding commands and config option isn't really that important. seems to work fine in Chrome!

closed 2017-09-11 by MightyPork