espterm/espterm-firmware #142: Non-block cursors leave ghosts when moving

MightyPork, 2017-09-11:
this happens when i move left/right (hold the cursor key). Cursor need not be blinky for this to happen. Block cursor leaves only green "ghosts" in the debug overlay (what is it?), the small cursors leave also their clones.

[image]

[image]

cpsdqs, 2017-09-11:
### The Hitchhiker's Guide to the Debug Overlay
The debug overlay—used for debugging and looking like you're a hacker. Even in its yet short lifetime, three great myths about it have already taken the masses: some say the mysterious green cells have been updated and marked as requiring a redraw by the almighty diffing algorithm; others say the ominous magenta cells were redrawn; and cpsdqs says cells with red outlines have wide characters.

I think this bug is being caused because I'm not actually storing whether or not something was drawn to have a cursor

cpsdqs, 2017-09-11:
Should be fixed in https://github.com/espterm/espterm-front-end/commit/5d2e8d845a86fc56b320c1aad461a8fecf61816a

MightyPork, 2017-09-11:
looks good. but there's a new funny bug - when the cursor is right from two blink-style characters (also somewhere below and right above?), it thinks it should blink (not with the characters, but on it's own, with the slow cursor timer)

[image]

MightyPork, 2017-09-11:
oh, rather, it seems to sometimes forget about its own blink enable flag (and sometimes think it's true when it's off)

cpsdqs, 2017-09-11:
that's actually intentional; the cursor renderer swaps the foreground and background colors, basically inverting the cell (but not the way the invert blend mode does)

MightyPork, 2017-09-11:
idk ... it does not blink at all when cursor blinking is enabled and there are no blinky characters

cpsdqs, 2017-09-11:
oh I should be moving the cursor blink thing to before the didUpdate. Should be fixed in https://github.com/espterm/espterm-front-end/commit/c82a955dfcbc7567c6b1109835096a1da26318fb

MightyPork, 2017-09-11:
just one more nitpick - switching from style "line" to "bar" produces this until the cursor moves
[image]

cpsdqs, 2017-09-11:
https://github.com/espterm/espterm-front-end/commit/f74e689196c890d23c163919baf34e69c4794bf3

MightyPork, 2017-09-11:
:+1: works perfectly

closed 2017-09-11 by MightyPork