espterm/espterm-firmware #143: Automatically enable / disable FancyGraphics™ based on screen content

MightyPork, 2017-09-11:
there's no need to redraw or even check cursor neighborhood if we know there are no wide/tall characters on the screen at all

(tall characters may be tricky to detect, but i think they're gonna be rare. maybe 𝄞 ? could have a list of the more common ones...)

a similar mechanism for counting can be used as is now employed to count blinky cells

cpsdqs, 2017-09-11:
Should we even consider a situation where the user has set the font to Zapfino
[image]

MightyPork, 2017-09-11:
i have something very unpleasant to say to such an user

cpsdqs, 2017-09-25:
So this is the only thing left that's priority—fancy graphics is still really slow for full screen redraws, and there's an abundance of tall characters; even # is taller than the cell size (by, like, a pixel) in whatever font Safari chose.
I think the goal should rather be to speed up fancy graphics than to hack it by only enabling it if necessary, while it's still really slow; here's some options:
- Find a more efficient way of clipping regions (e.g. whatever algorithm voxel renderers use to reduce triangle counts)
- Cache character bitmaps—not really sure about this one. Might speed up all rendering
- This option only listed for completeness: use OpenGL (requires font cache)
- other ideas?

MightyPork, 2017-09-25:
I think the only case where this matters are symbols / emoji that go outside the box horizontally.
⌛↯☢, weather symbols (for meteostation UI) ...

maybe I could add a checkbox to permanently enable it if people want it? And i guess we can move it out of priority… that's quite niche

cpsdqs, 2017-09-25:
So, none of the above
works for me, less work yay

MightyPork, 2017-09-25:
i mean if you get some genial idea to make it faster, that would be nice :man_shrugging:

cpsdqs, 2017-09-25:
That's my ideas up there xD

cpsdqs, 2017-10-27:
number three at ~20ms per draw
[image]

MightyPork, 2017-10-27:
if it's faster than what we had and compatible with what we currently support ¯\_(ツ)_/¯
but 20ms quite a lot, no?

MightyPork, 2017-10-27:
110 minified without demo? :eyes:

cpsdqs, 2017-10-27:
yeah, non-prod is 145k xD

cpsdqs, 2017-10-27:
I'll push it to a new branch

MightyPork, 2017-10-27:
mmk but i don't think we should merge that in this state