espterm/espterm-firmware #146: Optionally do not intercept F5, F12 and F11, interferes with browser shortcuts

MightyPork, 2017-09-11:


cpsdqs, 2017-09-11:
Also ⌥⌘I (dev tools)

MightyPork, 2017-09-11:
I don't think we capture _that_?

cpsdqs, 2017-09-11:
No, but it does capture the ⌥I part of it, which means it sends ˆ every time

MightyPork, 2017-09-11:
hnnnn

there are various options in the keymaster config, so if you wanna play with that ...
non-mac shortcuts are far easier to fix

MightyPork, 2017-09-11:
I've implemented this for F5, F11 and F12 now. do you want to try that �I ?

MightyPork, 2017-09-11:
maybe this is related to it always binding ctrl+letter and also command+letter

```
 /** Bind a keystroke to message */
  function bind (combo, str) {
    // mac fix - allow also cmd
    if (combo.indexOf('ctrl+') !== -1) {
      combo += ',' + combo.replace('ctrl', 'command')
    }
```

so maybe that's not the right behavior

closed 2017-09-12 by MightyPork