espterm/espterm-firmware #147: More terminal config options

cpsdqs, 2017-09-11:
Currently, these options can't be configured:
- font family
- font size
- line spacing/character spacing (aka `gridScaleX`/`Y`)
   - should definitely be live-updating

MightyPork, 2017-09-11:
live updating I think is not important, you do not normally have the config page open at the same time as the main screen. There could be a little preview canvas on the settings screen?

Then it's just a matter of strategically placing %replacements% in term.php for the initial load

cpsdqs, 2017-09-11:
Preview canvas would work too (and it can replace the theme color preview as well)

cpsdqs, 2017-10-28:
[image]

(I attempted it but it's really hard to write code that I can't compile in a language I never use)

MightyPork, 2017-10-30:
family and size are now configurable