espterm/espterm-firmware #152: Add keyboard shortcut bar

cpsdqs, 2017-09-11:
Most soft keyboards don't have arrow keys, escape, tab, ctrl, alt, shift, meta, etc. and using a terminal is rather unwieldy, so there should be an optional keyboard extension bar that'd sit on the screen bottom and provide those keys, like e.g. here

[image]

MightyPork, 2017-09-11:
there's an app for this called Hacker Keyboard for Android

cpsdqs, 2017-09-11:
on iOS there isn't

MightyPork, 2017-09-11:
well you know who's implementing that them :p
maybe this could be combined with the debug bar idea :thinking:

cpsdqs, 2017-09-23:
Added a basic keyboard bar (only on iOS) directly below the screen (can't use `position: fixed` because that's just shoved under the keyboard)
[image]
[image]

I also moved the fit-to-screen button above the screen (and forgot to say so in the commit). This is because for whatever reason page scrolling does not work at all on an iPhone, so you can't scroll down to even press the fit-to-screen button, though horizontal scrolling works. Moving it up allows access to most or all of the page depending on the screen height

MightyPork, 2017-09-24:
moving that button up is a good addition! I think this can be closed, seems to work ok

closed 2017-09-24 by MightyPork