espterm/espterm-firmware #160: Vertical lines on light bg with chrome on Windows

MightyPork, 2017-09-11:
probably some nonsense with rounding again.
I haven't tried this, they're screenshots I received

one idea to try as a work-around could be to change the back-plane color based on the predominant background color, if there is one

[image]

MightyPork, 2017-09-11:
I can kinda reproduce this by zooming in chrome (and firefox)

cpsdqs, 2017-09-11:
also happens in fullscreen mode on Android. I'm not sure what's causing it though

MightyPork, 2017-09-11:
i've added a work-around that sets canvas bg to the most common bg of all cells, should do the trick for now

closed 2017-09-11 by MightyPork