espterm/espterm-firmware #162: Weird gap under canvas

MightyPork, 2017-09-11:
got this after removing pretty much all styles - the gap is still there.

any ideas where it's coming from?

[image]

cpsdqs, 2017-09-11:
I don't see anything (the yellow part is the recolored `#screen` container)
[image]

MightyPork, 2017-09-11:
the padding is even on all sides. Or, well. so the stylesheet says and was the case with legacy screen

you also have it wider at the bottom

cpsdqs, 2017-09-11:
oh that's the CSS line-height. `<canvas>` is an inline block so it adds that

MightyPork, 2017-09-11:
Oh. that's stupid. and dev tools don't highlight that

closed 2017-09-11 by MightyPork