espterm/espterm-firmware #170: Make ⌘V use slow pasting/bracketed pasting

cpsdqs, 2017-09-13:
iTerm2:

[image]

This would require doing all input through something like the soft keyboard input which supports the `paste` event (the terminal should probably have an unfocused style, too)

MightyPork, 2017-09-13:
if you want "slow pasting", the text upload tool already implements that, with delay settings and everything

cpsdqs, 2017-09-14:
yeah, point was that pasting normally would also use it

cpsdqs, 2017-09-16:
After some testing I found out:

- clipboard events (`copy`, `cut`, `paste`) work fine without inputs
- clipboard events do not work if you call `e.preventDefault()` on `keydown` events
- keymaster calls `e.preventDefault()` on `keydown` events
- ∴ clipboard events do not fire

Possible solutions:

- [ ] don't fix this and let pasting just send a bajillion key events
- [ ] write something like keymaster that's less messy and make that not block clipboard events

MightyPork, 2017-09-16:
It's perfectly okay to hack the library I think, if that helps

MightyPork, 2017-09-16:
I've already modified it anyway (was missing some function keys)

cpsdqs, 2017-09-16:
well I would hack the library but I can't deal with code that indecipherable xD
...well, I can, but it'd take forever. It took me an hour to figure out the clipboard event situation.

MightyPork, 2017-09-16:
i sense a new side project coming :P

cpsdqs, 2017-09-16:
well, I did more or less unintentionally make a side branch by unifying input events (such that all input goes through the soft keyboard input and the terminal border dims when it's not focused) thinking it would enable input events
but very well, I shall write a keyboard event manager
...actually, I might just refactor term_input and put it in there

MightyPork, 2017-09-16:
i was thinking if you made it work as a js library that's better than keymaster (bc it kind of sucks for some things) that might be quite popular

cpsdqs, 2017-09-16:
nah, this is just tedious, not complicated; and all the specific features required here would need to be generalized to everything and that's too much work and I can always migrate later

MightyPork, 2017-09-16:
riight
well i would still hack keymaster but you like your es6, so go ahead :P

cpsdqs, 2017-09-16:
Added clipboard shortcuts (https://github.com/espterm/espterm-front-end/commit/97e08e71bd19ee61330b2ee15fc22c0ccf1edfa4), and on macOS ⌘C (which isn't overridden by the key bindings) actually works as expected. Pasting currently just uses `s[string]`.
Next part: TermUpl and the clipboard should share the same pasting API lest it be duplicate code.

MightyPork, 2017-09-16:
there are some quite important settings in the upload tool tho. eg. if you paste a long file without delays, it (a) can hog the server, and (b) the target application may not keep up

so i propose the upload dialog opens automatically when you try to paste a multiline (or longer than X) text

MightyPork, 2017-09-16:
ofc you dont see this without the hardware. should I mail you one esp8266? :P

cpsdqs, 2017-09-16:
I feel like there's more to it than just taking an ESP8266 and connecting it to USB? xD
Also, what should be max length for normal pasting? And should normal pasting be bracketed?

MightyPork, 2017-09-16:
I'm working with [this one](http://www.ebay.com/itm/LoLin-V3-NodeMcu-Lua-WIFI-Development-Board-/112218985271?epid=728639520&hash=item1a20c5d337:g:gvgAAOSwo4pYO~-K), so. yeah, just plug it in USB :P

and the flashing tool works on mac i think. building the firmware is a bit trickier ...

All pasting should be bracketed when bracketed paste is enabled (the application asks for it)
and the Long Length .. like, idk really

the paste tool splits long lines by 100 chars (?), you can't send more than that in one go or the websocket gets all messed up (idk what's going on there, it's written in C and full of pointers)

cpsdqs, 2017-09-16:
it is done and it works https://github.com/espterm/espterm-front-end/commit/9dee9e16286e39a17c44c31b66bb84b4622d44ee
(probably)

MightyPork, 2017-09-16:
will test that tmw but looking at the code, lgtm

cpsdqs, 2017-09-18:
no comments probably means no issues; hereby closed

closed 2017-09-18 by cpsdqs

MightyPork, 2017-09-18:
no comments mean I forgot about it :P

MightyPork, 2017-09-18:
whoa whoa it works

i can even paste with middle-click, ctrl-shift-v and shift-insert

cpsdqs, 2017-09-18:
huh. Blame your browser xD

MightyPork, 2017-09-18:
oh no that's the intended behavior. I'm surprised it works so well

what doesn't work is ctrl+insert that should copy (that i think is trivial to add, but not really important), and copy-on-select to X clipboard, but that's probably impossible with javascript (and of course shift-delete that should Cut)

cpsdqs, 2017-09-18:
yeah can't really cut from a terminal unless you want me to delete those screen cells xD
and if you want copy-on-select you can call `copySelectionToClipboard` in `selectEnd`, but I feel like that should be an opt-in feature

MightyPork, 2017-09-18:
ah no that would't be right

there are two clipboards on Linux, the X clipboard that uses the middle button and select, and the ctrl+c,v,x one. i think the insert combos manipulate the latter, but I'm not entirely sure, i dont use those normally.

I often have stuff in both at once, it's really handy! and there is also clipboard history, of course