espterm/espterm-firmware #171: See-through gaps between cells background on some systems

MightyPork, 2017-09-13:
partly mitigated by setting background color to the most common cell bg, but it's still visible in some cases, like this:

[image]

if this gets fixed we can also remove the background color counting which certainly ain't making the load routine any faster

cpsdqs, 2017-09-14:
The bug is caused due to the devicePixelRatio being an ugly value like 1.7999999523162842 (at e.g. 90% zoom) and because I set that as the transform and do nothing else, it's technically the browser's fault. Ways to fix it:

- ~round the DPR to a nice value (e.g. 1.8)~ ✘ still happens
- ~handle scaling manually~ ✘ still happens (or messes with redrawing)
- round the DPR to a very nice value (e.g. 1, 1.5, 2) and scale the canvas to fit ✓

Needs testing though

MightyPork, 2017-09-14:
Guenter reported it happens also without zoom, so that's not quite adding
up.

If you make some test page to do measurements (and like, put the results in
a textarea to copy), I'll send him a link

cpsdqs, 2017-09-14:
some screens have a 1.5 or 1.75 DPR by default so that might be it. And what would I measure anyway? The issue is that the browser's ctx.setTransform doesn't really work well for weird floaty values, and since I set the transform to the DPR, that should be the only relevant variable here. Though since I forgot to remove code I used for debugging, red lines will seep through the gaps if the bug should still happen

MightyPork, 2017-09-14:
fixed in 1.1.2

closed 2017-09-14 by MightyPork