espterm/espterm-firmware #175: Use rectangles to draw half- and full-block characters

MightyPork, 2017-09-15:
because e.g. my firefox uses some garbage font and it ends up like this:

[image]

this would be nice to also have for line drawing characters, but that seems to work more reliably than the half-blocks.

cpsdqs, 2017-09-15:
What about box drawing/shading characters (▒)?

MightyPork, 2017-09-15:
if you think that's doable, that would also be nice, but i thought it'd be too expensive

cpsdqs, 2017-09-15:
nooo text rendering is the part that's most expensive; if we used bitmap fonts this'd render the whole screen in less than ten milliseconds

MightyPork, 2017-09-15:
well talking of that, I would totally dig this font from the old IBM PC
[image]
but it doesnt have almost any unicode so that'd look awful when mixed

cpsdqs, 2017-09-15:
Terminus?

MightyPork, 2017-09-15:
idk i don't really like terminus. maybe profont (but that also doesn't cover much)
objectively the ibm font is quite ugly, but _r e t r o_

maybe we could steal the front from cool-retro-term

MightyPork, 2017-09-15:
ummm okay so it is using terminus xD

butthe font is 300 kB

cpsdqs, 2017-09-15:
Actually if you gzip it it's 50kB, which is... still huge
If you use only one style, it's 16kB gzipped, which is... still huge
You could of course make a fancy standalone client but I feel like that's not within the scope of this project

MightyPork, 2017-09-15:
16 kB is not so bad, there is still a couple hundred kB left in the flash of the smallest module.
Bigger issue is the strain loading multiple files puts on the processor (that's why I put the Fontello font as inline data into the stylesheet). The browser goes berserk and tries to load all at once and there are max 4 connection slots available.

MightyPork, 2017-09-15:
As for making a stand-alone client, once I add telnet, the whole browser thing can be switched off and people can use iTerm or whatever on their desktop

cpsdqs, 2017-09-15:
Added most block drawing characters (except the quadrants) in https://github.com/espterm/espterm-front-end/commit/0520b043f8c7fa88f2947fe0c2c117593782aab0

MightyPork, 2017-09-15:
works nicely! I'll add the remaining quadrants, just because

cpsdqs, 2017-09-15:
so
most quadrants won't draw for me 
[image]

MightyPork, 2017-09-15:
lolz

but that shouldn't happen? it has that cache busting thing already added in
[image]

closed 2017-09-15 by MightyPork

MightyPork, 2017-09-15:
fullscreen redraw with the dense shading character takes 100 ms. hopefully people don't do that a lot xD

cpsdqs, 2017-09-15:
if that becomes an issue, we'll switch to https://developer.mozilla.org/en-US/docs/Web/API/CanvasRenderingContext2D/createPattern

MightyPork, 2017-09-15:
hmm it even has a pretty decent support

MightyPork, 2017-09-15:
btw putting a block cursor over the block characters produces some hilarious animations