espterm/espterm-firmware #177: Don't show a gray screen forever if the server doesn't accept the connection

cpsdqs, 2017-09-15:
Currently, it'll just show this and reload the page every few seconds
[image]

cpsdqs, 2017-09-15:
In conjunction with this, it may be good to have a better indicator for “you've been disconnected” than the flickering gif in the top-right corner which goes off-screen when you scroll

MightyPork, 2017-09-15:
Yeah some overlay with "Reconnecting(flashing dots)" would be nice
The whole reconnect thing is a little rough atm,I think one of the timers
sometimes fires when it should be stopped and reloads it when it's no
longer needed

cpsdqs, 2017-09-15:
I added this 
[image]

MightyPork, 2017-09-15:
is it in git? I still have the old behavior when the esp restarts

cpsdqs, 2017-09-15:
it is now

[image]

MightyPork, 2017-09-15:
what a mess xD
it works, with some glitches:
- the regular ajax spinner is still popping up with each request
- it flashes for like 2 segments and then hides again during the heartbeat polling

cpsdqs, 2017-09-15:
well the code for setting the status screen is in term.js:9–14, change it all you want

cpsdqs, 2017-09-15:
So about something related to this issue; why exactly does it _reload_ when the ping is successful?

MightyPork, 2017-09-15:
i think it's from back when the initial done load was done via ajax and not by sending `i` to the WS.
now naturally the better solution would be to just re-open the socket

MightyPork, 2017-09-15:
reloading the page also re-loads the window title that's atm not sent back with the initial load (`i`). but that not happening is a minor drawback

closed 2017-09-15 by MightyPork