espterm/espterm-firmware #179: No way to copy text without a CMD button

MightyPork, 2017-09-15:
context menu is good for nothing and ctrl-c is consumed by the input handler (if it would do anything anyway)

[image]

MightyPork, 2017-09-15:
at least I _think_ you can copy on mac. certainly not on linux

cpsdqs, 2017-09-15:
I thought I added ⌃⇧C?

MightyPork, 2017-09-15:
ah yes cs-c works. we need to document that
a hack for the context menu would be nice (but it's also kinda cool you can save the image?!)

closed 2017-09-15 by MightyPork

cpsdqs, 2017-09-15:
here's a proof of concept for the hack https://codepen.io/cpsdqs/pen/eGNogw?editors=0010
it's really hacky.

MightyPork, 2017-09-15:
> here's a proof of concept for the hack https://codepen.io/cpsdqs/pen/
> eGNogw?editors=0010
> it's really hacky.

whoa that's some serious clickjacking. let's maybe not

cpsdqs, 2017-09-15:
Yeah we could also just use a custom context menu (could keep the “save image”)

MightyPork, 2017-09-15:
if there can be both that's a big win imo- but [low prio]