espterm/espterm-firmware #180: Draw line drawing characters with lines or rects

MightyPork, 2017-09-15:
maybe it would look nice, idk about the speed and also it'll be a pain cause there are over 100 of them

mind, we don't need all of them for sure (namely I don't care about the halflings)
```
        0  1  2  3  4  5  6  7  8  9  A  B  C  D  E  F
0x2500  ─  ━  │  ┃  ┄  ┅  ┆  ┇  ┈  ┉  ┊  ┋  ┌  ┍  ┎  ┏
0x2510  ┐  ┑  ┒  ┓  └  ┕  ┖  ┗  ┘  ┙  ┚  ┛  ├  ┝  ┞  ┟
0x2520  ┠  ┡  ┢  ┣  ┤  ┥  ┦  ┧  ┨  ┩  ┪  ┫  ┬  ┭  ┮  ┯
0x2530  ┰  ┱  ┲  ┳  ┴  ┵  ┶  ┷  ┸  ┹  ┺  ┻  ┼  ┽  ┾  ┿
0x2540  ╀  ╁  ╂  ╃  ╄  ╅  ╆  ╇  ╈  ╉  ╊  ╋  ╌  ╍  ╎  ╏
0x2550  ═  ║  ╒  ╓  ╔  ╕  ╖  ╗  ╘  ╙  ╚  ╛  ╜  ╝  ╞  ╟
0x2560  ╠  ╡  ╢  ╣  ╤  ╥  ╦  ╧  ╨  ╩  ╪  ╫  ╬  ╭  ╮  ╯
0x2570  ╰  ╱  ╲  ╳  ╴  ╵  ╶  ╷  ╸  ╹  ╺  ╻  ╼  ╽  ╾  ╿
```

cpsdqs, 2017-09-15:
Best case scenario:

- one function for drawing arms up/left/right/down with a thin stroke, bold stroke or double-struck stroke
- draw double-struck strokes such that their ends overlap seamlessly with both double-struck arms and join with thin stroke arms
- for 0x2500–0x250B use ((codePoint - 0x2500) / 2) % 2 to decide direction

cpsdqs, 2017-09-17:
So I tried writing some code for this, and I have to ask: how big can it be? Because the amount of code to render all that is—while it does have a repeating pattern—a lot

MightyPork, 2017-09-17:
if it's over say 50 kB minified, that wouldn't be ideal, but there is still space left.

I'd write it like

```
drawLines(top, right, button, left) { // <- eg. 0 - none, 1 - single, 2 - thicc, 3-double
  a bunch of horrible ifs
}
```
and then a switch that calls drawLines for each point ...

MightyPork, 2017-09-17:
if you gzip the output folder (without the humongous sourcemap), + around 240 kB of code, it should ideally fit in 500 kB

technically there is 1 MB available, but if I ever want to implement OTA update, only half is available

cpsdqs, 2017-09-17:
Added: https://github.com/espterm/espterm-front-end/commit/a79b08a16bfe5f3be7e97b78a3d39c05db247751

However, there's one remaining issue; the characters in the 0x2552–0x256C are slightly wrong (right is unicode):
[image]

cpsdqs, 2017-09-17:
...also, term_screen.js is now a whopping 1758 lines

MightyPork, 2017-09-17:
it looks like trash in my chrome
[image]

MightyPork, 2017-09-17:
I'll try to move this onto a branch so I can make a release even if it's not finished

MightyPork, 2017-09-17:
there, it's on branch `box-drawing`.
also I force-pushed to `work` so pls update your ref

cpsdqs, 2017-09-17:
Oh dear, @‌1x ruins everything

MightyPork, 2017-09-17:
maybe adding +0.5 everywhere can fix that

cpsdqs, 2017-09-17:
nope, then it'd just break if your system has a different font

MightyPork, 2017-09-19:
i think i'll try to redo this using rects, they worked nicely for the block chars

cpsdqs, 2017-09-19:
hm, maybe? You might want to round the center so that it’s on a whole-pixel value if the DPR is 1 or lower