espterm/espterm-firmware #183: Split the monstrous screen class to smaller units

MightyPork, 2017-09-17:
but I'm not sure how ...

for start maybe we can at least have stuff like notification showing, beeping and setting labels elsewhere (ie. the entire thing that branches by the first letter of the update object)

cpsdqs, 2017-09-17:
well it would certainly be nice if there were a way to not have to put everything external into global scope

MightyPork, 2017-09-17:
It goes through babel already, so if you think that's a way to make it cleaner...

MightyPork, 2017-09-17:
or well that webpack thing. if the overhead is not too big

cpsdqs, 2017-09-17:
about 2kb minus babel? (I didn't check how much webpack itself was)

MightyPork, 2017-09-17:
i'll make an issue for that (bc i like issues :)

MightyPork, 2017-09-18:
okay so I've split off themes and the color table for a start 
(and the `themes` module is for now in global scope for use by the color previews)

cpsdqs, 2017-09-20:
this commit https://github.com/espterm/espterm-front-end/commit/9ccf9dd2cf1920fd143e199476efbd667175ac94

MightyPork, 2017-09-20:
that's a big one

i'll try to test it asap

closed 2017-09-24 by MightyPork