espterm/espterm-firmware #184: Check if Webpack is feasible / try to use it

MightyPork, 2017-09-17:


MightyPork, 2017-09-17:
really though I have 0 idea how this works, other than having used it when hacking on mastodon

cpsdqs, 2017-09-17:
So webpack works fine on term.tpl, but all the other pages have HTML4-style inline scripts that use whatever they want so they break. What should be done about those?

MightyPork, 2017-09-17:
I think this mostly boils down to `$` and `tr` and the `WiFi` object. Can they be injected to the global scope?

wificonn should be self-contained iirc

MightyPork, 2017-09-17:
hmmmm app.js is now 93 kB vs 51 kB?

MightyPork, 2017-09-17:
oh nvm that must be the demo. But it's still 17 kB larger

MightyPork, 2017-09-17:
some observations: webpack's own overhead is about 4 kB (could be less if it didn't put tons of useless comments everywhere), the rest is due to worse minified libs?

cpsdqs, 2017-09-17:
oh yeah I totally forgot to minify it

cpsdqs, 2017-09-17:
So the 51 kB is outdated; the `work` branch is actually 68 kB. Webpack, minified, is 60 kB. I have no idea how that works.

MightyPork, 2017-09-17:
right you are
can you push the minification? i'll try to get the other pages to work

MightyPork, 2017-09-17:
did i mess something up?
[image]

cpsdqs, 2017-09-17:
""works as intended""
fixed xD

MightyPork, 2017-09-17:
looks like all works, but phpstorm decided it now doesn't like chibi and the short functions in utils.js (`qs`, `qsa`, `mk`) and marks everything as "not found" :/

closed 2017-09-17 by MightyPork