espterm/espterm-firmware #187: Add a list of supported browsers to the readme

MightyPork, 2017-09-18:
- chrome
- chromium (naturally)
- opera (bc it's really just chrome)
- brave (ditto)
- firefox
- safari
- konqueror (against all odds)
- edge? i havent tried that

cpsdqs, 2017-09-18:
edge works but—at least in BrowserStack—it's incredibly slow

cpsdqs, 2017-09-18:
Installed Windows ten; with Firefox and Edge: Firefox works fine, but Edge is reeeeally laggy. Currently figuring out why.
Apparently drawing takes ~780ms. It's absolutely ridiculous .-.

MightyPork, 2017-09-18:
90 ms per cell? WHAAAAAT

well i dont think we can expect much from microsoft but that's pretty shitty

is it in a VM? I had some bad luck with windows 10 in virtualbox

MightyPork, 2017-09-18:
we can just tell people not to use it :3

cpsdqs, 2017-09-18:
Okay, here's a full report of the `draw` method:
- setup: <1ms
- generate update map: 60ms
- generate redraw map: 80ms
- draw backgrounds: 260ms
- draw characters: 380ms

MightyPork, 2017-09-18:
that's perfectly horrible

let's just say we don't support IE I guess

MightyPork, 2017-09-18:
btw microsoft owns https://modern.ie

....

yeah

cpsdqs, 2017-09-18:
An IE-compatible version goes well beyond 376 kB with all those polyfills lol
So what's the list of unsupported browsers, to show a warning for?

MightyPork, 2017-09-18:
a warning? ... i mean, why not. looks like it sucks in any flavour of IE / Trident / Edge and Opera Mini.
It _might_ work with android browser, i can't test that. Probably wont work on blackberry also

we can also have a Cookie Notice after I implement session cookies LOL

the supported browser list is just so I can point people to it when they ask

closed 2017-09-18 by MightyPork