espterm/espterm-firmware #188: custom escape sequences to play musical notes

MightyPork, 2017-09-18:
hmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmmm

cpsdqs, 2017-09-18:
well, as long as you have ram left
a probably non-exhaustive list of note properties:
- frequency
- duration
- wave type
- harmonics ([see `PeriodicWave`](https://developer.mozilla.org/en-US/docs/Web/API/PeriodicWave))
- decay
- reverb
- volume
- stereo L/R

MightyPork, 2017-09-18:
err I think some simple beeping would be enough (pitch, duration)
back-end code is easy, just the B message would be expanded with beep type