espterm/espterm-firmware #190: Demo doesn't work: Uncaught TypeError: t.child.off is not a function

MightyPork, 2017-09-23:
[image]

[image]

[image]

also looks like it doesn't draw the whole background, just pieces (must be because of the crash)

cpsdqs, 2017-09-23:
Fixed https://github.com/espterm/espterm-front-end/commit/91c17e0fc38ab634f3ce7fd890a54c08dee72b7e

closed 2017-09-23 by cpsdqs

MightyPork, 2017-09-23:
> Fixed espterm/espterm-front-end@91c17e0

That was quick

cpsdqs, 2017-09-23:
Also fixed the background issue. For somer reason, the palette is switched from 0 to 1 and then 0 again halfway through, and I also wasn't clearing the drawnScreen when the palette is switched. It now renders correctly but the screen flashes black for a moment now