espterm/espterm-firmware #193: Selecting is messed up

MightyPork, 2017-09-24:
idk what exactly happened there yet, probably some logic that assumes fg and bg hold the current color but they can be 0,0 when bits 8 and 9 (HAS_FG, HAS_BG) are not 1

[image]

cpsdqs, 2017-09-24:
aaah what happened espterm-emulator is no longer compatible

cpsdqs, 2017-09-24:
Fixed—you accidentally made it look up the selection colors from the palette https://github.com/espterm/espterm-front-end/commit/a196f9a9de50010604ad3cbd00e775f1b8f30a7d

closed 2017-09-24 by cpsdqs

MightyPork, 2017-09-24:
ah yeah i broke it real good lol
but it shan't happen again at this scale, utf8 data fields can stretch to 2.5 bytes