espterm/espterm-firmware #197: try to use passive event listeners

MightyPork, 2017-09-24:
not sure if this helps anything, but chrome is complaining about those with verbose logging enabled

[image]

cpsdqs, 2017-09-24:
Can't make `wheel` passive because passive event listeners can't preventDefault. Also why are we using the non-standard `mousewheel`?

MightyPork, 2017-09-24:
alright, just reporting what i see :p

i'm recycling what i used earlier, so it must be from some old project ...

cpsdqs, 2017-09-25:
There's not a lot of event listeners listed in that violation message that can be passive
in fact, there's only one
https://github.com/espterm/espterm-front-end/commit/16436a9a9886a692803785bfc000ee721d7d7a7b

closed 2017-09-26 by MightyPork