espterm/espterm-firmware #215: API for controlling ESP8266 GPIOs remotely

MightyPork, 2017-09-27:
toggle pins, set multiple at once, read pins, read ADC...

schneemaier, 2018-01-28:
Would be really useful feature. Today i would be able to use it to reset my remote mini computer, instead of waiting till tomorrow evening when i get there :)

schneemaier, 2018-05-24:
Any new release expected in the near future which could include this feature?

schneemaier, 2018-05-24:
Nice one! Will look into more detail

schneemaier, 2018-08-07:
I know it is summertime :) but any chance that there will be update on this feature in the the next 4-6 weeks or a feature to do ota update of the firmware?

Thanks

MightyPork, 2018-08-13:
I've fallen victim to Another Side Project (https://datatable.directory/), but I didn't forget. I have this bad habit of having too many new projects and too little time. I hope to get to it soon :`]

MightyPork, 2018-08-18:
Okay, I decided to give this a try this weekend. So far it looks like I'll be able to provide API to control GPIO4, GPIO5 and GPIO16. @schneemaier is configuring the initial levels after boot important to you? If not (all go to 0 at startup), this could be added without any config or UI changes at all

schneemaier, 2018-08-18:
Thanks for the try. Unfortunately I use ESP8266-01 board and that only has gpio 0 and 2 available. As I don't use gpio2 debug output hoped for using that...

Output level is important in the final application as it controls the reset pin and it need to be high. (If it is not possible to control the initial state I can use additional transistor to invert it)

MightyPork, 2018-08-18:
okay, I'll see what can be done about that. 

It should be possible, just count with a burst of UART output on GPIO2 during boot, even if you set it as GPIO, that can't be easily prevented

closed 2018-08-18 by MightyPork

MightyPork, 2018-08-19:
Now we have an API and also a GUI page:
https://github.com/espterm/espterm-firmware/releases/tag/2.4.0-pre2