espterm/espterm-firmware #216: Recolor screen margin

cpsdqs, 2017-09-28:
Especially noticeable in this screenshot:
[image]
The screen margin doesn't update its color with the screen backgrounds or themes and can thus kind of break the seamless style.

Possible solutions:
- [ ] Don't fix it at all
- [ ] Set it to the default background and be done with it
- [ ] Set it to the average background color
- [ ] Set it to the adjacent cell background color (ComputerCraft does this, or at least it did last time I used it which is like 2015)

MightyPork, 2017-09-28:
idk ... muted adjacent color sounds fun

talking of changing color, there can also be an OSC to change the border color...

cpsdqs, 2017-09-28:
muted? The adjacent cell color thing meant that the cells along the perimeter would essentially just get bigger

MightyPork, 2017-09-28:
yeah I don't really like that

something like this:
[image]

cpsdqs, 2017-09-28:
could also do this with almost zero CSS hacks (at least in Safari, Chrome's backdrop-filter is still incredibly glitchy but behind a flag so no one will notice)
[image]

MightyPork, 2017-09-28:
that wouldn't be bad

cpsdqs, 2017-09-28:
Implemented https://github.com/espterm/espterm-front-end/commit/aa41cf512e03946212106af6afb3d974e362463d

closed 2017-09-28 by cpsdqs

MightyPork, 2017-09-28:
i'll have a look - after making back-end compile again xD

cpsdqs, 2017-09-28:
haha, blurry
[image]

MightyPork, 2017-09-28:
looks nice
but u may wanna fix that horrible overlapping triangle :p

cpsdqs, 2017-09-28:
you mean the powerline symbols? that's the font's fault

MightyPork, 2017-09-28:
idk what's a powerline but looks like it could be drawn using a polygon...
[image]

cpsdqs, 2017-09-28:
hm good idea, though I seriously doubt anyone would use vim with powerline (or in this case, airline) in ESPTerm xD

MightyPork, 2017-09-28:
I mean no, but they're nice symbols :P
I'll add them together with the block drawing characters to a new codepage