espterm/espterm-firmware #217: front-end bugs report for 2.0.0

MightyPork, 2017-09-28:
report from Guenter:

>On Android KB comes up automatically - no KB Icon click required
>Scaling works well on Android with different screens

>Tried it on a current Edge version - after some startup trouble with stylesheet Terminal screen looks ok
>Terminal setup page had no colors on Edge

cpsdqs, 2017-09-29:
They keyboard is supposed to come up when you tap on the screen—unless it comes up with no user action at all, that might be a bug. I can't reproduce the working OK part on Edge 15.15603; it's still extremely laggy for me. I can confirm the setup page has no colors on Edge; error: `Object doesn't support property or method 'forEach'`, which is referring to document.querySelectorAll

MightyPork, 2017-09-29:
Yeah the qsa thing we should fix, that's just a matter of using chibi loop
instead of vanilla

closed 2017-09-29 by MightyPork