espterm/espterm-firmware #221: Add some light themes

cpsdqs, 2017-09-29:
(custom themes would also work, but that's effort)

e.g.
- Tango Light
- default xterm theme
- Solarized Light

MightyPork, 2017-09-29:
umm we have light themes (and xterm default theme is dark???)
[image]

cpsdqs, 2017-09-29:
yeah, but every app still assumes that 0 is the background color. I guess this can be a wontfix.

Also
[image]

MightyPork, 2017-09-29:
huh? send a PR to your apps I guess...
dont really mind if you add more themes tho. blows up the code a bit but it's not something dramatic

MightyPork, 2017-09-29:
[image]

MightyPork, 2017-09-30:
i've added solarized light and high-contrast dark, add the others if you find them...

closed 2017-10-04 by MightyPork