espterm/espterm-firmware #222: blurry padding looks funny in corners

MightyPork, 2017-09-29:
[image]

cpsdqs, 2017-09-29:
Yeeeah that's because those are two divs; I don't really know how one would go about making that one xD

MightyPork, 2017-09-29:
it's not so bad without the blur tbh [image]

cpsdqs, 2017-09-29:
feel free to remove the backdrop-filter without the `-webkit-` prefix then

MightyPork, 2017-09-29:
do you think they could be bevelled like this? (idk how to do that)
[image]

cpsdqs, 2017-09-29:
Yes. It would be the hack of the century. But not the worst hack in this project :P

cpsdqs, 2017-09-29:
nevermind it doesn't work

MightyPork, 2017-09-29:
aww

cpsdqs, 2017-09-29:
okay I did make one that works
https://codepen.io/anon/pen/MEoGYP
but only in Safari .-.

MightyPork, 2017-09-29:
i had high hopes for that thick border hack but it blurs even under transparent :c

MightyPork, 2017-09-29:
chrome blurs even outside clip-path? wtf

closed 2017-09-30 by MightyPork