espterm/espterm-firmware #224: soft kb should not open on canvas tap if mouse reporting is enabled

MightyPork, 2017-09-29:
quoting

>On Android the soft KB comes up as soon as I touch the terminal window - no mouse event is generated
I tested this on a Tablet and a Phone - both with Android 6.01 & Chrome
AMIduos Android Emulator on W10 Surface Pro - same problem
This definitely worked in an older version
>
> Bug ->  the mouse function is broken on Android due to soft KB popping up
>
> I digged out my Surface Pro, which runs an older W10 and has a touchscreen
Chrome works with a mouse, but not on touch - other tabs works fine with touch in Chrome
Firefox works fine with touch & mouse
Edge is slow as hell, but displays ok -  works with touch
Soft KB only comes up when Icon is clicked - ok

(edge report checks out xD)

cpsdqs, 2017-09-29:
oh yeah I guess it shouldn't do that lol
I'll fix that after I do sass things

MightyPork, 2017-09-29:
okay cool
also any idea what's up with the chrome touch events he mentions?

cpsdqs, 2017-09-29:
well I don't really understand what “does not work on touch” is supposed to mean xD

MightyPork, 2017-09-29:
hes been testing the screen, I assume it's about mouse reporting not firing for touches
that being old surface with windows 10 leads me to believe the chrome isn't exactly 63, if that's any hint

closed 2017-09-29 by MightyPork