espterm/espterm-firmware #225: cursor is not redrawn/moved on screen when only "C" section present

MightyPork, 2017-09-29:
try eg moving cursor in a text editor, it gets redrawn only after you stop moving it

i think theres something messed up with the blink timeouts but havent relaly looked yet

MightyPork, 2017-09-29:
with the old format, always the entire screen was redrawn, must be related

cpsdqs, 2017-09-29:
Can't reproduce?

MightyPork, 2017-09-29:
I can 100% with latest `work`

make sure your messages have only C
[image]

MightyPork, 2017-09-29:
also its more prominent with a non-blinky cursor, blinky cursor tends to recover after about 0.5s

cpsdqs, 2017-09-29:
oh yeah the cursor update thingy doesn't schedule a draw. Fixed now

closed 2017-09-29 by cpsdqs

MightyPork, 2017-09-29:
awesome, works fine now!