espterm/espterm-firmware #228: Add simple front-end(s)

MightyPork, 2017-09-30:
from another Guenter's testing report, the renderer / socket doesn't work with these:
- Iphone 5 IOS with Chrome
- "rather new LG SmartTV with WebOS"

WebOS seems to use some old version of WebKit, iPhone probably uses whatever Apple Webkit engine was available at the time (that's 5 years ago if the OS wasn't updated :frowning:)

>I was originally hoping that we get a rather universal WEB frontend, which works even with older devices
>This would enable the reuse of old tablets and phones as display devices
>
>Now I see that this is a very complex subject with lots of dependencies on OS and Browser - I underestimated the work a lot...
>
>Is there any way to have a "simple" mode which works in a limited fashion with older browsers and smart TVs ??
>
>I could imagine just a simple semistatic 8 color  HTML page which displays the content and refreshes every x seconds
>this would be good enough for displaying status & measurement values - but not for real interactive use
>Maybe as automatic fallback if the "server cannot connect " status appears

This isn't super important, but if we add it it could have some potential.

**Implementation ideas:**

- Mouse input isn't really that important, if we have the fixed buttons
- There can be a text field with a button for input (avoids using JS events, real-time typick would be annoying if the screen wouldn't update to show the new characters anyway)
- Screen can be drawn as styled spans in a `<pre>` block. I don't think it's a big deal if that breaks with wide glyphs. Or it can be without styles entirely
- We can use `<b> <i> <u>` for attributes.
- *Colors* either as inline css (but that sucks w.r.t. data size), completely left out, or via css classes (should be fine if we give up on themes and use just CGA)
- Alternatively it can be simply *black&white* (colors are nice but bold and italic works too)

The re could be two "dumbness tiers" to this:

- *Fully static*, possibly with meta refresh or JS refresh (latter is better so we can suppress it while user is typing in the text box)
- *Simple with JS and XHR* (no sockets or canvas)

Another really cool thing to add: **Log Viewer** front-end
- We have a 10 kB large data array for the screen content. If I make it an union with a long char array, that's a massive 10 kB buffer for backlog. This could be very useful for certain classes of applications

**Sub-tasks**
- [ ] implement simple front-end
- [ ] add simple serialization routine and CGI endpoint to back-end
- [ ] implement log view (this cannot co-exist with the normal or simple front-end)
- [ ] add URLs to access those new front-ends
- [ ] add config option to select any as default

cpsdqs, 2017-09-30:
on an iPhone all browsers must use WebKit—would be nice to know which iOS
version though. I have to ask though, what exactly is ESPTerm for anyway? xD

MightyPork, 2017-09-30:
i asked for the versions.

now what it's for: ???

- ok, primarily it's a education platform, students from this embedded class will use it to connect to their arduino-like projects on a breadboard with a phone.
- you can also stick it anywhere together with some arduino or similar to have remote interface. I can see people using this for home automation, diy meteo station monitoring, stuff like that

really any kind of embedded diy hacks could be augmented by adding this...

cpsdqs, 2017-09-30:
hm okay, as long as you have RAM left lol

MightyPork, 2017-09-30:
won't be an issue, the CGI and "filesystem" lives in flash. we're at 330 kB, I want to fit under 500 kB so that's still almost 200k left. and those things wont really use any extra RAM (the log buffer will share memory with the screen array)

MightyPork, 2017-09-30:
update on the affected ios version - it's IOS 8.1.3 (12B466)

cpsdqs, 2017-09-30:
yeah pretty much, I think they just let the certificates expire
[image]

MightyPork, 2017-09-30:
Wow that's quite shitty (not that I was surprised)

MightyPork, 2017-10-07:
Adding front-end to the list:
SVG uploaded via the serial port. Can be streamed via websocket...

this will allow fancy graphs etc