espterm/espterm-firmware #230: Button count

cpsdqs, 2017-09-30:
Everything works fine if the button count (specified in the protocol) is 5 or higher. However, if it's lower, everything breaks because the parser iterates over the button elements. What exactly is the button count supposed to do? Add/remove buttons, or just aid in avoiding unnecessary data?

MightyPork, 2017-09-30:
I want to, sometime in the future, to add the possibility to change the button count (i.e. js would build the buttons dynamically). I've added it to the payload to avoid format changes when it's implemented

cpsdqs, 2017-09-30:
Alright, client-side is done then https://github.com/espterm/espterm-front-end/commit/3fba2d88c40c7fc78998a2b52d3e7268a52a1474
(I think this issue can be closed ¯\_(ツ)_/¯)

closed 2017-09-30 by cpsdqs

MightyPork, 2017-09-30:
yeah, i'll make a new ticket for the back-end and config