espterm/espterm-firmware #232: Add a non-English locale to test if i18n works

MightyPork, 2017-09-30:


cpsdqs, 2017-09-30:
It works but some words are clipped
[image]

MightyPork, 2017-09-30:
*ahem* maybe don't have such long words :P

MightyPork, 2017-09-30:
okay so the label width could of course be increased, but some things are just ridiculous:

[image]

can some of the longest labels be shortened? I'll then make a DE binaries with the next release (and also CS)

cpsdqs, 2017-09-30:
uhh, not really, but I guess you could truncate

MightyPork, 2017-09-30:
gave u some more room there
also we now have Czech, yay!
[image]

closed 2017-09-30 by MightyPork