espterm/espterm-firmware #236: Color triangle does not work depending on screen Y

cpsdqs, 2017-09-30:
It's really strange; sometimes it works, and then you scroll down or up a bit and it hits a threshold where it just doesn't do anything when you click and drag. Tested in Safari. Also what is that sorcery—SVG with a pixelation filter or something? xD
It can't be inspected, it excuses itself and disappears when you open dev tools :(

MightyPork, 2017-09-30:
yeah it's somewhat buggy - I've converted the library to ES6, but the bugs survived apparently (if not multiplied)

i think it's not doing a v good job at figuring out the relative mouse coordinate

MightyPork, 2017-09-30:
oh and it's a canvas

MightyPork, 2017-09-30:
wow what a shitshow

it measures the element offset on load and then thinks it's fixed forever (?)

MightyPork, 2017-09-30:
removed like 20 lines of javascript and used getboundingclientrect instead, seems to work ok

closed 2017-09-30 by MightyPork