espterm/espterm-firmware #241: Some weird back rectangle next to screen

MightyPork, 2017-10-03:
[image]

I haven't reproduced this yet, got it in a report. No idea wtf it is

cpsdqs, 2017-10-03:
any other details?

MightyPork, 2017-10-03:
Not really, I'm asking them to inspect it. Added the chrome version... It's
latest 'work'

MightyPork, 2017-10-03:
It's all really odd, I'm not near the pc to check it atm but I've never seen anything like that
[image]
[image]
[image]

MightyPork, 2017-10-03:
(he tried deleting some stuff, for whatever reason)

cpsdqs, 2017-10-03:
Only reason I can think of is that the blur on the screen margins is messing up the renderer

MightyPork, 2017-10-03:
We could make it toggleable xD
I really don't get why it's all so weirdly asymmetrical tho. There
shouldn't be any dark/light region around the screen (or to the side)...

cpsdqs, 2017-10-03:
No we should know if it's actually causing it before doing anything like that lol

MightyPork, 2017-10-03:
ikr lol
I didn't see any such bug yesterday when I tested that release...
And it's not some cached asset thing either, there's a cache busting fix
already

MightyPork, 2017-10-03:
got a report that this can be fixed by adding the background color to body
it doesn't really make sense, but if it works...

MightyPork, 2017-10-03:
i can't reproduce it and we have a work-around added, so i'll close this

closed 2017-10-03 by MightyPork