espterm/espterm-firmware #244: Demo bugs

MightyPork, 2017-10-04:
[image]
also the color is gray by default after this, should be 7 or default..
- local echo lacks Exit button

- typing sudo auto-completes "github" immediately, no way to enter the sandwich command
- make also auto completes github as the target
- same for the "theme" command

possible improvements>
themes command output is imo quite overwhelming and hardly gives you an idea of how the themes feel
mouse command doesnt showcase that we can also track mouse motion and drag/drop

cpsdqs, 2017-10-04:
espterm/espterm-front-end@fcfd351
I added the exit command, fixed an issue with autocomplete appearing when it shouldn't, but I don't get what you mean by “auto-completes "github" immediately”

MightyPork, 2017-10-04:
well I typed "make " and it started typing github and then executed it (or
something)

You can see it on the demo page

cpsdqs, 2017-10-04:
can't reproduce—the only reason why it would do that is if you pressed the button below

MightyPork, 2017-10-04:
huh that's odd. Maybe the button was focused and space pressed it, that's
pretty sucky

cpsdqs, 2017-10-04:
Here's a fancier mouse demo
[image]

cpsdqs, 2017-10-04:
I don't know what to do about `themes`, any ideas?

MightyPork, 2017-10-04:
looks great!!

MightyPork, 2017-10-04:
idk.

one idea - show a fullscreen preview and let user flip through them, "q" for exit or something
but it'd be nice to have their names in the themes array for that... they're now only in the html select box

MightyPork, 2017-10-04:
a thing about this theme preview tho ... the 0-15 themes play some role, but even greater effect have the default colors. there are far too many combinations to just show a big preview of everything

cpsdqs, 2017-10-04:
I'll tackle that tomorrow then. If I'm feeling exceptionally bored I might accidentally add an RGB color picker inside the terminal

MightyPork, 2017-10-04:
just please don't accidentally add sixel lol

cpsdqs, 2017-10-04:
if I do all the code will be grafted onto the renderer in the demo.js file xD

closed 2017-10-04 by MightyPork