espterm/espterm-firmware #245: switchable languages........

MightyPork, 2017-10-04:
if we exclude the help page, that's like 35 kB uncompressed per language, not a huge deal - we just must figure out how to have multiple languages in js

MightyPork, 2017-10-04:
presumably the active language could be defined in a similar way to `_demo` or `_root`

cpsdqs, 2017-10-04:
the webpack loader can easily load multiple languages—I have no idea about the PHP though. Maybe the server could perform substitutions instead of having a bunch of duplicate data?

MightyPork, 2017-10-04:
it could. i'm not sure that's worth it, as it'd require exporting the constants to C and building some horrible look-up table with an x-macro. it could be really slow...

MightyPork, 2017-10-04:
oh, and the help page is pre-gzipped, so we can't do any replacements on that (i.e. the menu)
which .. uhhhhhh is a problem. might need to figure out some js hack with a get parameter

MightyPork, 2017-10-04:
hmmpf

let's maybe not, for now. although japascript _is_ on every page afaik

cpsdqs, 2017-10-04:
well, if you do decide to do this, use the smallest method lest it take up like 100Mb

MightyPork, 2017-10-04:
i hope your js translation plan is something more sophisticated than what my router is using

[image]

[image]

MightyPork, 2017-10-04:
they certainly don't give a damn about file size there with keys like this
[image]

cpsdqs, 2017-10-04:
before I accidentally express my opinion on that code I'll just answer the question with a ‘yes very much so’ xD