espterm/espterm-firmware #246: possibility to have an image background

MightyPork, 2017-10-04:
> a background picture from some external server - giving a room layout map or piping diagram
URL should be definable dynamically by a controlmessage  from the micro

The idea is to load an image behind the canvas as a background.

@cpsdqs do you think the renderer can be augmented to optionally don't draw default background and instead leave it transparent?

cpsdqs, 2017-10-05:
technically you could even draw the background with canvas, yes

cpsdqs, 2017-10-06:
I don't know why one would want this but it's possible now ¯\\\_(ツ)_/¯
[image]

MightyPork, 2017-10-06:
the "why" is simple :P

- seen those? this is for the colours

[image]

ok, a more likely example:

[image]

cpsdqs, 2017-10-07:
huh fair enough. Well, if you add it to the update string you can now just set `termScreen.backgroundImage` to the URL and everything should work fine (might need some CSS to fix scaling though)

closed 2017-10-07 by cpsdqs