espterm/espterm-firmware #249: Color settings for control buttons

plachta11b, 2017-10-18:
Feature request: Add color changeable buttons in Terminal settings->Initial Settings->button colors for games in terminal. It can be useful for some applications. What about ability to add user CCS inside settings? This feature could bring interesting hacks for custom projects.

MightyPork, 2017-10-18:
good idea, i think we should add this

cpsdqs, 2017-10-18:
Client-side support added: https://github.com/espterm/espterm-front-end/commit/ce46ad7e963ff072e35acca1ae1604ccb0c91702

MightyPork, 2017-10-18:
kinda. i think this should be appended to the buttons update topic, it'll usually change together with the labels

cpsdqs, 2017-10-18:
Yeah you can just split each entry by `\x02` or just have the array repeat again for the color values or something and have the color there (you do need a null value though)
Also, if we add a user CSS feature, we could also consider adding user JS as well (and also add an `espterm` wrapper object or something in the global scope)

MightyPork, 2017-10-18:
how about each button text field starts with the color (two words, like the bg and fg colors) followed by the text and terminated by 0x01?

I don't have storage for user CSS, the entire RW area is 2kB large and already half used (though i imagine it could be expanded if need be)

cpsdqs, 2017-10-18:
Sure, as long as it has a null color

MightyPork, 2017-10-18:
null color? does that mean "use default"?

cpsdqs, 2017-10-18:
yes (null as in `button.style.background = null`)

MightyPork, 2017-10-18:
? [image]

cpsdqs, 2017-10-18:
(what are you asking? xD)

MightyPork, 2017-10-18:
still not sure what you mean by null background ...

cpsdqs, 2017-10-18:
I meant you set the style attribute's background to null such that it falls back to whatever you set in the stylesheet

MightyPork, 2017-10-18:
...hmm
there can be nothing in the stylesheet, if we always send a color --- and support 24bit

MightyPork, 2017-10-30:
[image]

closed 2017-10-30 by MightyPork