espterm/espterm-firmware #251: Does not work with port redirection

schneemaier, 2017-10-22:
I use port redirection on my router to access different services remotely and stunnel for https on my home network. Example redirect port 4000 from the external interface to port 80 via stunnel to reach the espterm. But this does not work, I get server connection failed message from the terminal.

MightyPork, 2017-10-23:
I think we'll need more details to debug this.

> I get server connection failed message from the terminal
is that a response from ESPTerm, it's output to the debug pin, or something else?

If you build the firmware with -DDEBUG_HTTP=1, it'll log any received connections, alternatively you could debug this with WireShark in promiscuous mode (or locally, if you can run it on the gateway).

I'm not familiar with stunnel, maybe if you tell me how to set it up, i'll be able to reproduce it

cpsdqs, 2017-10-23:
I just noticed an odd detail—https? Modern browsers block connections to any insecure protocol if the page was loaded via https, and ESPTerm uses ws

MightyPork, 2017-10-23:
that's a good observation, sounds like the socket isn't correctly forwarded in this setup. I don't think there's anything we can do on our end - except fallback to ajax polling, but that will be horribly laggy and stress the server.

MightyPork, 2017-10-23:
hmm :thinking: 

it's occured to me we have 'http://' hard-coded in some places, and also I'm not sure what will 'ws://' do if the page is loaded over https ...

[image]

schneemaier, 2017-10-24:
All,

here are some more info and comments from my setup

> I think we'll need more details to debug this.
> 
> > I get server connection failed message from the terminal

> is that a response from ESPTerm, it's output to the debug pin, or something else?
>
> If you build the firmware with -DDEBUG_HTTP=1, it'll log any received connections, alternatively you could debug [...]

This is a message i got in the browser window
[image]

About stunnel: stunnel is TLS proxy service which provide encryption support for devices, services which does not support it out fromthe box. https://www.stunnel.org/index.html I use it with several services (like mjpg-streamer). 

This is the stunnel config file:
`root@abc:~$ cat /etc/stunnel/ser.dd.a.conf `
`pid=/tmp/ser2.pid`
`[ser.dd.a]`
`cert = /etc/letsencrypt/live/cb-cr.no-ip.org/fullchain.pem`
`key = /etc/letsencrypt/live/cb-a.no-ip.org/privkey.pem`
`sslVersion = all`
`client = no`
`accept = 4002`
`connect = ser.dd.a:80`

> it's occured to me we have 'http://' hard-coded in some places, and also I'm not sure what will 'ws://' do if the page is loaded over https ...

Would it be possible to implement a logic on the browser side: if page loded with https, then open socket with wss: instead of ws: ?

Hope this helps

cpsdqs, 2017-10-24:
<details>
<summary>Yes</summary>

https://github.com/espterm/espterm-front-end/commit/0794030210c5f268f879c1b82448f3b9525dc5f9

</details>

schneemaier, 2017-10-26:
Tried the new 2.3.0-pre1 and connection via https + port redirection with stunnel works nicelly. I will do more test later.

Thanks for the quick support!

MightyPork, 2017-10-26:
good to hear, closing

closed 2017-10-26 by MightyPork