espterm/espterm-firmware #257: ubuntu 14.4 resolved issues

wolfm2, 2018-01-14:
I spent the last couple days getting it to work.  Here are the results:

Upload an (any) ssh id to github and checkout with:
git clone  --recursive git@github.com:espterm/espterm-firmware.git

The stock node and yarn in the repos **will never work**  you must install the newest binaries from:
https://nodejs.org/en/download/package-manager/#debian-and-ubuntu-based-linux-distributions
https://yarnpkg.com/lang/en/docs/install/info 

Yarn pounds on DNS when you do "yarn install" in the front-end dir.  If your router can't keep up and you get:
  **There appears to be trouble with your network connection. Retrying..**.
Add this to your hosts file: 151.101.12.162 registry.npmjs.org

PHP5 will have to be installed too.  The repo version worked fine.

Once installed and you attach to the AP, you can test it using the same serial adapter you flashed with:
minicom -D /dev/ttyUSB0 -b 9600   # <- change speed as appropriate

Other (non?) issues:
  The app webpage reloads **_fast_**.  If your link to the esp AP is slow, the page will reload several times before you get all the needed files.  I had several black screens & un-cssed white screens with gui elements before it loaded properly.    Would it make sense to make this timeout longer?

  I didn't know much about node and may never modify the front-end code.  It would have saved a day of work/research/installs if you committed pre-processed front end files somewhere for those who just want to modify the c code.  

  Given that extra non-embedded tools are required (yarn, node, php) a bullet-point list in the top .md file of all the infrastructure (more than the SDK) the project requires and what it does would be very useful.

Thanks for the code!
M

MightyPork, 2018-01-14:
Good to hear you got it working!

If you want to add some installation notes to the front-end repo README, I'll gladly accept a PR (or even to this project's README).

As for the reload timeout, you can adjust those in the JS files (I don't remember the exact place, but the structure/naming is quite readable so you should find it). I never had any issues with speed though (using a garbage router I got from the ISP)

I'm working on a different project (for my thesis), so ESPTerm is experiencing a bit of a hiatus, I won't have time to work on it until June. Feel free to send PRs if you make any useful improvements!

cpsdqs, 2018-01-14:
(ftr the refreshing code is actually here and not in a JS file https://github.com/espterm/espterm-front-end/blob/master/pages/term.php#L9)

MightyPork, 2018-01-14:
also here https://github.com/espterm/espterm-front-end/blob/master/pages/term.php#L84

and then there's also the heartbeat timer, somewhere

wolfm2, 2018-01-14:
Yeah.  I'm guessing that my speed issue is that I live in New York city an I can often see (no joke) 60+ 2.4ghz routers in my apartment.  More than half of them have good signal quality.   My sad esp is probably drowning in all the packet collisions.

M