espterm/espterm-firmware #260: connection to esp-AP laggy when looking for other network

plachta11b, 2018-05-12:
I tried to setup connection to local lan over my access point and ESPTerm works well. ESPTerm start to lag after I turn of that external local lan AP. ESPTerm is looking for this network so forcefully that connection over esp-AP become unstable. Probably ESPTerm should give up (or slow down) after some time. Cause of this problem can be hard to find.

there is ping output if that can somehow help
https://pastebin.com/n4GytpFE

MightyPork, 2018-05-12:
This would be helpful.... I didn't find a way to configure the re-connect interval in the API; this happens automatically in the SDK, outside our control

MightyPork, 2018-05-12:
btw. as a workaround, I implemented that "hold BOOT (until LED starts flashing) to reset to AP" feature