espterm/espterm-firmware #261: Failed to clone / init submodules

schneemaier, 2018-09-05:
Any idea why i get the below error when it trye to clone or intit the submodules?

Tried both
git clone --recursive https://github.com/espterm/espterm-firmware.git (failed)
and 
git submodule init (success)
git submodule update (failed)

`schnee@DellUbuntu:~/Workspace/espterm-firmware$ git submodule update
Cloning into '/home/schnee/Workspace/espterm-firmware/front-end'...
Warning: Permanently added the RSA host key for IP address '192.30.253.112' to the list of known hosts.
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:espterm/espterm-front-end.git' into submodule path '/home/schnee/Workspace/espterm-firmware/front-end' failed
Failed to clone 'front-end'. Retry scheduled
Cloning into '/home/schnee/Workspace/espterm-firmware/libesphttpd'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:MightyPork/libesphttpd.git' into submodule path '/home/schnee/Workspace/espterm-firmware/libesphttpd' failed
Failed to clone 'libesphttpd'. Retry scheduled
Cloning into '/home/schnee/Workspace/espterm-firmware/front-end'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:espterm/espterm-front-end.git' into submodule path '/home/schnee/Workspace/espterm-firmware/front-end' failed
Failed to clone 'front-end' a second time, aborting`

MightyPork, 2018-09-05:
The submodules are linked using the SSH address (e.g. `git@github.com:espterm/espterm-front-end.git`), maybe you didn't add your SSH public key to GitHub?

I recall someone else had this problem before, maybe you could find it in the bugtracker. A workaround, other than simply adding your key, is to change the URLs in `.git/config` to the https format, like `https://github.com/espterm/espterm-front-end.git`

schneemaier, 2018-09-05:
Thanks! That helped

closed 2018-09-05 by schneemaier